damellis/attiny #54: Merge the 1.5/1.6 version into master.

cmaglie, 2015-02-10:
Since 1.6.0 is out now, it makes sense now to merge the 1.5 branch into master.

NicoHood, 2015-02-10:
+1
But please first have a look at those PRs

https://github.com/damellis/attiny/pull/33

https://github.com/damellis/attiny/pull/30

damellis, 2015-02-13:
For the moment, I created branches ide-1.0.x and ide-1.6.x. In editing the tutorial, it seemed clearest to have the version number in the filename, as opposed to using "master" for 1.6 and "ide-1.0.x" for 1.0.x. I also kept master and ide-1.5.x as is in case anyone is using them. Once 1.6 has been out for a while and has more adoption, I'll delete ide-1.5.x and ide-1.6.x and move the 1.6.x code onto the master branch.

damellis, 2016-03-07:
Done.

closed 2016-03-07 by damellis