damellis/attiny #55: ATtiny Entry is not showing up in the Tools > Board menu of Arduino 1.0.6

Akfreak, 2015-02-25:
Sorry I am all new to this, I followed the tutorial at http://highlowtech.org/?p=1695 and the ATtiny Entry not showing up in the Tools > Board menu Arduino 1.0.6.

Here is a screenshot of of my prefs, and the location of the properly placed attiny folder inside the hardware subfolder http://gyazo.com/d76ff4d0e392ddec77e3e2375e28a263

Here is a screengrab showing ATtiny is not showing up in the Board Menu..http://gyazo.com/5e29a0d36271df6c1f0ea10a5d2d1752

I am going to try a reboot of my PC, but that doesn't seem right.

NicoHood, 2015-02-25:
As stated in the readme use this branch for 1.0.6 or use the newer IDE 1.6
https://github.com/damellis/attiny/tree/ide-1.0.x

Akfreak, 2015-02-25:
I feel so dumb, it was late and I saw 1.0.6. as 1.6. Thanks for pointing out my mistake

NicoHood, 2015-02-26:
Please close the issue :)

closed 2015-02-27 by damellis