damellis/attiny #56: Upload Error: "avrdude: stk500_recv(): programmer is not responding"

pediRAM, 2015-04-05:
Hi!
Hope someone has an Idea why it doen't work for me: compilation is ok, but uploading fails on original Arduino UNO (R3), Windows 7 (64 bit).

I've followed the instructions, step by step and built the circuit exactly as shown at:
"http://highlowtech.org/?p=1695", except my capacitator is 10uF 63V (instead of 10uF 25V).

What I've already tested:
Arduino ist working fine (tested several progs.)
ATtiny45 is new out of shop (has never been in use before).
Changed strips and checked connection by multimeter (connection-beeper)

You can see the full protocol of compilation and upload below.
Thank you in advance for any suggestions.
Best Regards.
**********************************************************
Arduino: 1.6.1 (Windows 7), Platine: "ATtiny, ATtiny45, 1 MHz (internal)"

Build-Optionen wurden verändert, alles wird neu gebaut

D:\Programme\Arduino-1.6.1/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=attiny45 -DF_CPU=1000000L -DARDUINO=10601 -DARDUINO_attiny -DARDUINO_ARCH_AVR -ID:\Programme\Arduino-1.6.1\hardware\arduino\avr\cores\arduino -ID:\Daten\Projekte\Arduino Projekte\hardware\attiny\avr\variants\tiny8 C:\Users\johndoe\AppData\Local\Temp\build6809648053743465599.tmp\Blink.cpp -o C:\Users\johndoe\AppData\Local\Temp\build6809648053743465599.tmp\Blink.cpp.o 
D:\Programme\Arduino-1.6.1/hardware/tools/avr/bin/avr-gcc -c -g -Os -w -ffunction-sections -fdata-sections -MMD -mmcu=attiny45 -DF_CPU=1000000L -DARDUINO=10601 -DARDUINO_attiny -DARDUINO_ARCH_AVR -ID:\Programme\Arduino-1.6.1\hardware\arduino\avr\cores\arduino -ID:\Daten\Projekte\Arduino Projekte\hardware\attiny\avr\variants\tiny8 D:\Programme\Arduino-1.6.1\hardware\arduino\avr\cores\arduino\hooks.c -o C:\Users\johndoe\AppData\Local\Temp\build6809648053743465599.tmp\hooks.c.o 
D:\Programme\Arduino-1.6.1/hardware/tools/avr/bin/avr-gcc -c -g -Os -w -ffunction-sections -fdata-sections -MMD -mmcu=attiny45 -DF_CPU=1000000L -DARDUINO=10601 -DARDUINO_attiny -DARDUINO_ARCH_AVR -ID:\Programme\Arduino-1.6.1\hardware\arduino\avr\cores\arduino -ID:\Daten\Projekte\Arduino Projekte\hardware\attiny\avr\variants\tiny8 D:\Programme\Arduino-1.6.1\hardware\arduino\avr\cores\arduino\WInterrupts.c -o C:\Users\johndoe\AppData\Local\Temp\build6809648053743465599.tmp\WInterrupts.c.o 
D:\Programme\Arduino-1.6.1/hardware/tools/avr/bin/avr-gcc -c -g -Os -w -ffunction-sections -fdata-sections -MMD -mmcu=attiny45 -DF_CPU=1000000L -DARDUINO=10601 -DARDUINO_attiny -DARDUINO_ARCH_AVR -ID:\Programme\Arduino-1.6.1\hardware\arduino\avr\cores\arduino -ID:\Daten\Projekte\Arduino Projekte\hardware\attiny\avr\variants\tiny8 D:\Programme\Arduino-1.6.1\hardware\arduino\avr\cores\arduino\wiring.c -o C:\Users\johndoe\AppData\Local\Temp\build6809648053743465599.tmp\wiring.c.o 
D:\Programme\Arduino-1.6.1/hardware/tools/avr/bin/avr-gcc -c -g -Os -w -ffunction-sections -fdata-sections -MMD -mmcu=attiny45 -DF_CPU=1000000L -DARDUINO=10601 -DARDUINO_attiny -DARDUINO_ARCH_AVR -ID:\Programme\Arduino-1.6.1\hardware\arduino\avr\cores\arduino -ID:\Daten\Projekte\Arduino Projekte\hardware\attiny\avr\variants\tiny8 D:\Programme\Arduino-1.6.1\hardware\arduino\avr\cores\arduino\wiring_analog.c -o C:\Users\johndoe\AppData\Local\Temp\build6809648053743465599.tmp\wiring_analog.c.o 
D:\Programme\Arduino-1.6.1/hardware/tools/avr/bin/avr-gcc -c -g -Os -w -ffunction-sections -fdata-sections -MMD -mmcu=attiny45 -DF_CPU=1000000L -DARDUINO=10601 -DARDUINO_attiny -DARDUINO_ARCH_AVR -ID:\Programme\Arduino-1.6.1\hardware\arduino\avr\cores\arduino -ID:\Daten\Projekte\Arduino Projekte\hardware\attiny\avr\variants\tiny8 D:\Programme\Arduino-1.6.1\hardware\arduino\avr\cores\arduino\wiring_digital.c -o C:\Users\johndoe\AppData\Local\Temp\build6809648053743465599.tmp\wiring_digital.c.o 
D:\Programme\Arduino-1.6.1/hardware/tools/avr/bin/avr-gcc -c -g -Os -w -ffunction-sections -fdata-sections -MMD -mmcu=attiny45 -DF_CPU=1000000L -DARDUINO=10601 -DARDUINO_attiny -DARDUINO_ARCH_AVR -ID:\Programme\Arduino-1.6.1\hardware\arduino\avr\cores\arduino -ID:\Daten\Projekte\Arduino Projekte\hardware\attiny\avr\variants\tiny8 D:\Programme\Arduino-1.6.1\hardware\arduino\avr\cores\arduino\wiring_pulse.c -o C:\Users\johndoe\AppData\Local\Temp\build6809648053743465599.tmp\wiring_pulse.c.o 
D:\Programme\Arduino-1.6.1/hardware/tools/avr/bin/avr-gcc -c -g -Os -w -ffunction-sections -fdata-sections -MMD -mmcu=attiny45 -DF_CPU=1000000L -DARDUINO=10601 -DARDUINO_attiny -DARDUINO_ARCH_AVR -ID:\Programme\Arduino-1.6.1\hardware\arduino\avr\cores\arduino -ID:\Daten\Projekte\Arduino Projekte\hardware\attiny\avr\variants\tiny8 D:\Programme\Arduino-1.6.1\hardware\arduino\avr\cores\arduino\wiring_shift.c -o C:\Users\johndoe\AppData\Local\Temp\build6809648053743465599.tmp\wiring_shift.c.o 
D:\Programme\Arduino-1.6.1/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=attiny45 -DF_CPU=1000000L -DARDUINO=10601 -DARDUINO_attiny -DARDUINO_ARCH_AVR -ID:\Programme\Arduino-1.6.1\hardware\arduino\avr\cores\arduino -ID:\Daten\Projekte\Arduino Projekte\hardware\attiny\avr\variants\tiny8 D:\Programme\Arduino-1.6.1\hardware\arduino\avr\cores\arduino\abi.cpp -o C:\Users\johndoe\AppData\Local\Temp\build6809648053743465599.tmp\abi.cpp.o 
D:\Programme\Arduino-1.6.1/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=attiny45 -DF_CPU=1000000L -DARDUINO=10601 -DARDUINO_attiny -DARDUINO_ARCH_AVR -ID:\Programme\Arduino-1.6.1\hardware\arduino\avr\cores\arduino -ID:\Daten\Projekte\Arduino Projekte\hardware\attiny\avr\variants\tiny8 D:\Programme\Arduino-1.6.1\hardware\arduino\avr\cores\arduino\CDC.cpp -o C:\Users\johndoe\AppData\Local\Temp\build6809648053743465599.tmp\CDC.cpp.o 
D:\Programme\Arduino-1.6.1/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=attiny45 -DF_CPU=1000000L -DARDUINO=10601 -DARDUINO_attiny -DARDUINO_ARCH_AVR -ID:\Programme\Arduino-1.6.1\hardware\arduino\avr\cores\arduino -ID:\Daten\Projekte\Arduino Projekte\hardware\attiny\avr\variants\tiny8 D:\Programme\Arduino-1.6.1\hardware\arduino\avr\cores\arduino\HardwareSerial.cpp -o C:\Users\johndoe\AppData\Local\Temp\build6809648053743465599.tmp\HardwareSerial.cpp.o 
D:\Programme\Arduino-1.6.1/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=attiny45 -DF_CPU=1000000L -DARDUINO=10601 -DARDUINO_attiny -DARDUINO_ARCH_AVR -ID:\Programme\Arduino-1.6.1\hardware\arduino\avr\cores\arduino -ID:\Daten\Projekte\Arduino Projekte\hardware\attiny\avr\variants\tiny8 D:\Programme\Arduino-1.6.1\hardware\arduino\avr\cores\arduino\HardwareSerial0.cpp -o C:\Users\johndoe\AppData\Local\Temp\build6809648053743465599.tmp\HardwareSerial0.cpp.o 
D:\Programme\Arduino-1.6.1/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=attiny45 -DF_CPU=1000000L -DARDUINO=10601 -DARDUINO_attiny -DARDUINO_ARCH_AVR -ID:\Programme\Arduino-1.6.1\hardware\arduino\avr\cores\arduino -ID:\Daten\Projekte\Arduino Projekte\hardware\attiny\avr\variants\tiny8 D:\Programme\Arduino-1.6.1\hardware\arduino\avr\cores\arduino\HardwareSerial1.cpp -o C:\Users\johndoe\AppData\Local\Temp\build6809648053743465599.tmp\HardwareSerial1.cpp.o 
D:\Programme\Arduino-1.6.1/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=attiny45 -DF_CPU=1000000L -DARDUINO=10601 -DARDUINO_attiny -DARDUINO_ARCH_AVR -ID:\Programme\Arduino-1.6.1\hardware\arduino\avr\cores\arduino -ID:\Daten\Projekte\Arduino Projekte\hardware\attiny\avr\variants\tiny8 D:\Programme\Arduino-1.6.1\hardware\arduino\avr\cores\arduino\HardwareSerial2.cpp -o C:\Users\johndoe\AppData\Local\Temp\build6809648053743465599.tmp\HardwareSerial2.cpp.o 
D:\Programme\Arduino-1.6.1/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=attiny45 -DF_CPU=1000000L -DARDUINO=10601 -DARDUINO_attiny -DARDUINO_ARCH_AVR -ID:\Programme\Arduino-1.6.1\hardware\arduino\avr\cores\arduino -ID:\Daten\Projekte\Arduino Projekte\hardware\attiny\avr\variants\tiny8 D:\Programme\Arduino-1.6.1\hardware\arduino\avr\cores\arduino\HardwareSerial3.cpp -o C:\Users\ [... cut ...]
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x03

avrdude done.  Thank you.

elpoolet, 2015-04-05:
Hi,

I can suggest you to use this tutorial : http://www.creativeapplications.net/tutorials/small-programmable-wearable-and-cheap-enjoying-the-attiny85-tutorial/

Cause I found your's not as complete than this.

The most important is to do the 2 steps : 
Burn bootloader and then program the ATTiny.

I tried this several times with a "chines copy" of an Arduino Uno R3 and all works fine.
I recommand to keep an Arduino as an ISP.

keep us nformed of your experimentations.

damellis, 2015-04-05:
pediRAM: the output you posted suggests that the computer (Arduino software) isn't able to find / communicate with the ArduinoISP sketch on the Arduino board. (That is, the problem seems to be happening before the Arduino board tries to communicate with the ATtiny.)

Did you upload the ArduinoISP sketch to the Arduino? Is COM2 the right port for your Arduino?

pediRAM, 2015-04-05:
Hello @elpoolet and @damellis!
You are right! 
I didn't upload the "ArduinoISP"-Sketch first... *facepalm*
Now after doing this step everything is working like a charm.
Thank you very very much for your Help! You guys saved my sunday!
Wish you nice/happy easter/sunday!
Kindly Regards.

closed 2015-04-05 by pediRAM

Prashanth2811, 2018-09-12:
Can you explain what do you exactly mean by uploading "ArduinoISP" sketch

Prashanth2811, 2018-09-12:
> Hello @elpoolet and @damellis!
> You are right!
> I didn't upload the "ArduinoISP"-Sketch first... _facepalm_
> Now after doing this step everything is working like a charm.
> Thank you very very much for your Help! You guys saved my sunday!
> Wish you nice/happy easter/sunday!
> Kindly Regards.

Can you explain what do you exactly mean by uploading "ArduinoISP" sketch

janpekar, 2018-09-23:
It means that you need to load File -> Examples -> ArduinoISP and upload it to your Arduino board which turns your connected Arduino into programmer device. After that you need to switch Tools -> Programmer -> Arduino as ISP and select proper board you want to program. If you don't know this and you don't want to use your arduino as programmer device for another avr, you are looking at wrong forum post and you have problem with usb/serial connection to your Arduino board.

Sakthivinnavan, 2020-03-24:
Hi i am using arduino uno board R3 model . i unexpectedly uploaded the code while bluetooth module conneted in D0 and D1 i got an error on uploading . after this i couldn't  upload any program
i am getting error message of following.....
Arduino: 1.8.12 (Windows Store 1.8.33.0) (Windows 10), Board: "Arduino Uno"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\hardware -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\hardware\tools\avr -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\sakth\OneDrive\Documents\Arduino\libraries -fqbn=arduino:avr:uno -vid-pid=1A86_7523 -ide-version=10812 -build-path C:\Users\sakth\AppData\Local\Temp\arduino_build_392843 -warnings=none -build-cache C:\Users\sakth\AppData\Local\Temp\arduino_cache_425298 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\hardware\tools\avr -verbose C:\Users\sakth\AppData\Local\Temp\untitled524745468.tmp\sketch_mar24a\sketch_mar24a.ino
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\hardware -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\hardware\tools\avr -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\sakth\OneDrive\Documents\Arduino\libraries -fqbn=arduino:avr:uno -vid-pid=1A86_7523 -ide-version=10812 -build-path C:\Users\sakth\AppData\Local\Temp\arduino_build_392843 -warnings=none -build-cache C:\Users\sakth\AppData\Local\Temp\arduino_cache_425298 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\hardware\tools\avr -verbose C:\Users\sakth\AppData\Local\Temp\untitled524745468.tmp\sketch_mar24a\sketch_mar24a.ino
Using board 'uno' from platform in folder: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\hardware\arduino\avr
Detecting libraries used...
"C:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\\hardware\\arduino\\avr\\variants\\standard" "C:\\Users\\sakth\\AppData\\Local\\Temp\\arduino_build_392843\\sketch\\sketch_mar24a.ino.cpp" -o nul
Generating function prototypes...
"C:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files\\WindowsApps\\ArduinoLLC.Ar [... cut ...]
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0xa5
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0xa5
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0xa5

avrdude done.  Thank you.

Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.

please help me to solve this problem