damellis/attiny #58: Add support for the ATmega168/328/644/1284

mschlenker, 2015-05-04:
Besides the name of this repo it would be nice to have support for the most common ATmegas that are available in breadboard friendly DIL package via your hardware definitions. I would volunteer by submitting pull request by pull request in turtle speed over the next few months.

damellis, 2015-05-05:
I'm still ambivalent about supporting ATmega's in this repository, in part because the tutorial / information about how to use them is somewhat different than that for the ATtiny. (They're much closer to the existing boards.) So the best thing might be for you to start by supporting these in a separate repository. Then we can decide later if it makes sense to merge them or not. (I do think it would be great to have good support for the ATmega on a breadboard processors, though, so I'd definitely encourage you to take this up!)

mschlenker, 2015-05-05:
OK, I started here: https://github.com/mschlenker/atmega-breadboard These are simply my HW definitions I assembled two years ago for the 1.5.x betas when I simply did not find anything for the new more flexible folder structure. I am going to remove the ATtinys from this repo and add the breadboard friendly ATmegas.

carlosefr, 2015-05-08:
I have been using these definitions for a while, for atmegas without a bootloader. It's based on the attiny configs.

https://github.com/carlosefr/atmega

damellis, 2016-03-07:
Closing in favor of Carlos's repository.

closed 2016-03-07 by damellis