damellis/attiny #59: Add package_attiny_index.json for boards manager support.

damellis, 2015-05-05:
So that people can automatically install the ATtiny boards from the boards manager.

per1234, 2015-05-08:
I have created a working implementation of installation of attiny to Arduino IDE 1.6.4 via Boards Manager. The URL is:
https://raw.githubusercontent.com/per1234/attiny/boards-manager-installation-test/package_damellis_attiny_index.json

I would like to create a pull request to add this capability to attiny but first I need the installation file to be available. The most common way of doing this seems to be a Github release. My installation file is here:
https://github.com/per1234/attiny/releases/download/v1.0.0/attiny.zip

Thanks, Per

damellis, 2015-05-08:
That's awesome! Thanks for helping with this.

What do you think about just linking to the zip for the latest commit, i.e. https://github.com/damellis/attiny/archive/42848608aa3a3712ccbb04109ef17bd49fc458b3.zip ? Is that going to have the right folder structure?

per1234, 2015-05-08:
Unfortunately it doesn't work because of the avr folder, everything needs to be moved up directly under the attiny folder.

damellis, 2015-05-08:
I made a new branch (https://github.com/damellis/attiny/tree/ide-1.6.x-boards-manager) and moved the files to the root folder. Does that look right? Can we use this file? https://github.com/damellis/attiny/archive/702aa287455f7e052cf94fd4949398fec0ef21b8.zip

per1234, 2015-05-08:
That works. I'll submit the pull request now.

damellis, 2015-05-08:
Thanks for your help!

closed 2015-05-08 by damellis