damellis/attiny #66: Not loading library

tytower, 2015-05-28:
Hi david Thanks for your contributions over the years . Helped me a lot 
I am getting an error loading arduino 1.6.4 and it will not now open at all.

 ./arduino
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
java.lang.NullPointerException
	at cc.arduino.contributions.packages.ContributedToolReference.resolve(ContributedToolReference.java:43)
	at cc.arduino.contributions.packages.ContributedPlatform.resolveToolsDependencies(ContributedPlatform.java:76)
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:92)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:615)
	at processing.app.Base.<init>(Base.java:306)
	at processing.app.Base.guardedMain(Base.java:222)
	at processing.app.Base.main(Base.java:133)
#

per1234, 2015-05-28:
Looks like maybe a similar issue to the one reported here: http://forum.arduino.cc/index.php?topic=324095.0
- Did you use Boards Manager to install attiny?
- Do you have any other Boards Manager URLs entered in ***File > Preferences > Additional Boards Manager URLs:***?
- Are you saying that your IDE will no longer load(only the splash screen shows)?

tytower, 2015-05-28:
The IDE no longer starts . There is no splash screen .The error is found by running the start script in a terminal.

I have ESP8266/Arduino in the boards manager also and it was loading without problem before I added attiny.

per1234, 2015-05-28:
That's the same combination that's causing anath3ma on the forum a problem. Are you using the ESP8266 URL:
http://arduino.esp8266.com/package_esp8266com_index.json ?

You should be able to get the IDE working by deleting the file Arduino15/package_esp8266com_index.json
Let me know if you can't find it and I'll help locate it.

I think the problem is with the ESP8266 file and not attiny because I had the same problem when I tried with ESP8266 and other URLs. I have successfully installed ESP8266 and attiny simultaneously using Adafruit's package index:
https://adafruit.github.io/arduino-board-index/package_adafruit_index.json
Which is based on the same project though it is esp8266 1.6.2 instead of 1.6.4 and esptool 0.4.2 instead of 0.4.4
So you could give that a try and I'll look at the other package index file and see if I can find the issue.

tytower, 2015-05-28:
No I went into /root/.arduino15 and deleted your json file for now as I am working on the ESP8266 module for now . Ill come back later for the tiny -thanks

   

https://adafruit.github.io/arduino-board-index/package_adafruit_index.json

per1234, 2015-05-28:
A workaround for having both is to do a manual installation of attiny and a Boards Manager installation of http://arduino.esp8266.com/package_esp8266com_index.json

per1234, 2015-05-29:
I have altered the attiny Boards Manager json index file to solve this issue. ESP8266 and attiny should now be able to be added simultaneously to your Additional Boards Manager URLs. Also, the problem with the Arduino IDE that was causing it not to load with the previous attiny json index file has been fixed and is available in the hourly build. Sorry for any inconvenience this has caused you.
Per

tytower, 2015-05-29:
Not a matter of inconvenience . Thank you for the library .Superb chip

damellis, 2016-03-07:
Sounds like this was fixed, so I'm closing this.

PR #67 (2015-05-29): Add tools field to package index
Missing tools field was causing incompatibility with other Boards Manager packages. Fixes #66. The Arduino IDE has been fixed to handle missing tools field by https://github.com/arduino/Arduino/commit/1b7574a8b9ac5cbbac4809bdd0a6a27014d7ea26.

Boards Manager URL for testing is: https://raw.githubusercontent.com/per1234/attiny/per1234-patch-add-tools/package_damellis_attiny_index.json
closed 2016-03-07 by damellis