damellis/attiny #68: Support for other fuse variants

CraigB1960, 2015-06-15:
I have been using Arduino 1.0.5 and had the following in the boards.txt file:
attiny85-8.name=ATtiny85 (CombinationLock)
attiny85-8.bootloader.low_fuses=0xF1
attiny85-8.bootloader.high_fuses=0xD5
attiny85-8.bootloader.extended_fuses=0xFE
attiny85-8.upload.maximum_size=8192
attiny85-8.build.mcu=attiny85
attiny85-8.build.f_cpu=8000000L
attiny85-8.build.core=arduino:arduino
attiny85-8.build.variant=tiny8

Your ATtiny85 supports works well, but does not have these particular fuses set.  Is there an easy way to put this board configuration into Arduino 1.6.4?

damellis, 2015-06-16:
You should be able to put a boards.txt file in your SKETCHBOOK/hardware directory. It will need to go in a sub-directory like CraigB/avr. (This is similar to 1.0.5, but with the addition of the avr/ folder, since 1.6.x supports multiple architectures.) You may need to add these to boards.txt:

attiny85-8.bootloader.tool=arduino:avrdude
attiny85-8.upload.tool=arduino:avrdude

And the platform.txt file here: https://github.com/damellis/attiny/blob/ide-1.6.x-boards-manager/platform.txt

CraigB1960, 2015-06-16:
Thank you!  Worked perfectly!

damellis, 2015-06-16:
Great!

closed 2015-06-16 by damellis