damellis/attiny #70: Jeelib cannot be used

stahlfabrik, 2015-06-22:
I need to have support for the library "jeelib", which can be downloaded here:
https://github.com/jcw/jeelib/archive/master.zip

When having this minimal sketch:
#include <JeeLib.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}

I get this error when selecting an Atiny84 (8 Mhz internal):

In file included from /Users/cstahl/Documents/Arduino/libraries/jeelib-master/JeeLib.h:17:0,
                 from sketch_jun22a.ino:1:
/Users/user/Documents/Arduino/libraries/jeelib-master/Ports.h:713:49: error: 'Serial' was not declared in this scope
     InputParser (byte size, Commands*, Stream& =Serial);
                                                 ^
/Users/user/Documents/Arduino/libraries/jeelib-master/Ports.h:714:60: error: 'Serial' was not declared in this scope
     InputParser (byte* buf, byte size, Commands*, Stream& =Serial);

When I select Arduino Uno for example, it compiles just fine. Also when I select the alternativ atiny core from google code.

But as THIS is the recommended third party core I want to use and embrace it! Any help greatly welcome! Love the board manager integration!

stahlfabrik, 2015-06-22:
The include statement got mangled. Of course I import Jeelib there!

damellis, 2015-06-22:
Looks like the library assumes that the Serial object exists, which corresponds to a hardware serial port. The ATtiny84 doesn't have a hardware serial port, so the object doesn't exist, giving that error. The code that uses the Serial object is only compiled for newer version of the Arduino core files, which is probably why you don't see the error on an alternative ATtiny core -- I think it includes an old version of the Arduino core files.

I'd suggest notifying the authors of the library about this. They should be able to modify the library not to rely on the Serial object if it doesn't exist. Unfortunately, I don't think there's a good way to solve this in these ATtiny files.

stahlfabrik, 2015-06-22:
Thanks for your reply! This alternative core brings its own Software Serial Code for serial output. And as you say, the Jeelib Library depends on the "Serial" being present. I actually started an issue over at jeelibs github to make them aware of the issue with THIS core (which is the recommended core according to arduino.cc). Because it is the recommended core I would like to work with it - but if the needed libraries are not compatible with it - I might fail in my project. I will see...

I wonder how the to find out, in which way your core and that other core maybe differ any further when it comes to Pin aliases etc...

damellis, 2015-06-22:
You can use the built-in SoftwareSerial library with this ATtiny core...  Right now, though, the library's code seems to assume that the hardware Serial object exists too. It should be possible to adjust that library so that it can work with SoftwareSerial with this ATtiny core, but I'm afraid I don't have time to help with that.

damellis, 2015-06-25:
I'm closing this issue because I don't think I can fix the problem within the ATtiny files. See: https://github.com/jcw/jeelib/issues/78 for more.

closed 2015-06-25 by damellis

JohnOH, 2015-08-10:
Hi @damellis is the TinyDebugSerial code present in earlier versions of the Tiny Core is still present in your version. It appears to me missing and the jeelib repository requires it.