damellis/attiny #72: Will not compile some libs due to missing definition of __AVR_ATtinyX5__

VaclavSynacek, 2015-06-25:
Manchester library (and maybe others too) will not compile for ATtiny85 due to due to missing definition of `__AVR_ATtinyX5__` . See http://arduino.stackexchange.com/a/1306 for more datails.

olvap, 2015-08-19:
any news on this?

lezhanger, 2015-09-24:
H, perhaps try this: https://github.com/mchr3k/arduino-libs-manchester/issues/13
I changed from AVR_ATtinyX5 should become AVR_ATtiny85 in the Manchester.cpp file and ATtiny85 now compiles.

psbs, 2015-10-09:
thanks a lot. it does compile on replacing  AVR_ATtinyX5 with AVR_ATtiny85  in the Manchester.cpp file

VaclavSynacek, 2019-08-10:
I do not need this any more and others seem happy with the workaround discussed => closing this 4 year old issue.

closed 2019-08-10 by VaclavSynacek