damellis/attiny #75: Can't compile after installing via board manager in 1.6.4 ide.

cb88, 2015-08-04:
C:\Program Files (x86)\Arduino-1.6.4\hardware\tools\avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=attiny45 -DF_CPU=1000000L -DARDUINO=10604 -DARDUINO_attiny -DARDUINO_ARCH_AVR -DSERIAL_BUFFER_SIZE 256 -IC:\Program Files (x86)\Arduino-1.6.4\hardware\arduino\avr\cores\arduino -IC:\Users\crayfield.MANTISSA\AppData\Roaming\Arduino15\packages\attiny\hardware\avr\1.0.1\variants\tiny8 C:\Users\CRAYFI~1.MAN\AppData\Local\Temp\build7322299124588225719.tmp\sketch_aug04a.cpp -o C:\Users\CRAYFI~1.MAN\AppData\Local\Temp\build7322299124588225719.tmp\sketch_aug04a.cpp.o 
avr-g++: error: 256: No such file or directory
Error compiling.

cb88, 2015-08-04:
I installed 1.6.5-r2 and it works with that version... Perhaps it is just a bug in 1.6.4.

damellis, 2016-03-07:
Closing since this seems to only affect old version of the IDE.

closed 2016-03-07 by damellis