damellis/attiny #76: TinyDebugSerial

JohnOH, 2015-08-10:
Hi @damellis. I hope this isn't duplicating but I noticed the original thread has closed.

Is the TinyDebugSerial code present in earlier versions of the Tiny Core still present in your version. It appears to be missing and the jeelib repository requires it.

damellis, 2015-09-08:
This core doesn't include TinyDebugSerial because the standard SoftwareSerial library works with it (although I realize it may be bigger or otherwise unsuitable for certain applications). I'm closing this because I'm not planning to include it. Ideally, it would be available as a library that could be used with a variety of cores.

closed 2015-09-08 by damellis

NicoHood, 2015-09-15:
I had a user in the IRC chat that complained about an issue with softserial. He was forced to use it ad baud 9600 which causes long ISRs. This then freezez the code too long and it crashes. I do not know of what softserial consists, but I guess it uses the usart like device that can at least add a bit automation to the usart interface.

A 2nd Library doesnt hurt, especially if its only used for the Attiny core, so why not ship this with the Attiny?

madmalkav, 2015-09-16:
Hi, I think NicoHood is refering to me. After some testing I have come to the conclusion that the problem of the serial not working is caused by the IR Remote library, I'm moving to IRLRemote. Thanks for your help guys.

NicoHood, 2015-09-17:
That might be a good idea ;) Also consider to use the PinChangeInterrupt with your Attiny ;)
But I think this still does not solve the main reason why your problem occured:
The Softserials ISRs take too long and this causes problems with the IRremote Interrupts. Using my library avoids this problem maybe, but the TinySerial would be still a better idea maybe. If you used AltSoftSerial both libraries could have used the same timers for interrupts.

madmalkav, 2015-09-17:
AFAIK AltSoftSerial isn't supported on attiny