damellis/attiny #78: arduino 1.6.6 board manager

piobel, 2015-08-20:
the json link don't work on board manager arduino 1.6.6

per1234, 2015-08-20:
I just tested it with Arduino IDE 1.6.6 2015/08/20 05:41 and it worked for me. Could you try again and give more information on what doesn't work please. If you have multiple Additional Boards Manager URLs in your preferences post the whole list.

per1234, 2015-08-21:
@piobel did you try to post some information as an attachment to the email? That doesn't work.

piobel, 2015-08-21:
please did you see the PNG attached ?

piobel, 2015-08-21:
HI,

reinstall IDE 1..6.5  an this is new error in board manager

per1234, 2015-08-21:
>please did you see the PNG attached ?

As I said before, **email attachments don't work on GitHub**. You need to either attach the image to a message on GitHub or type the error out in email.

damellis, 2016-03-07:
Not sure what's going on here so I'm closing this. If you continue to have problems, please re-open or create a new issue.

closed 2016-03-07 by damellis