damellis/attiny #81: Missing digitalPinToInterrupt definition

madmalkav, 2015-09-16:
The core doesn't define digitalPinToInterrupt for attinys.

NicoHood, 2015-09-17:
instead of -1 use NOT_AN_INTERRUPT
https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/variants/standard/pins_arduino.h#L61

eried, 2015-12-15:
I though the sale but it seems all pins can be interrupt in the tiny85. Does anyone have a working interruptions code with damellis/attiny?

NicoHood, 2015-12-15:
No this is wrong.

There are PinInterrupts (1 on Attiny I think) and PinChangeInterrupts (all pins on Attiny x5).

This code should work, but was never tested by me, at least not for Attinys, but for All other platforms:
https://github.com/NicoHood/PinChangeInterrupt

PinInterrupt issue still exist though, and should be added as well.

eried, 2015-12-15:
Thanks @NicoHood I'll try. I tried some other similar code but there was some incompatibility with SoftwareSerial.

NicoHood, 2015-12-16:
It is incompatible too. Otherwise you have to code your own interrupt vectors. Then you are working on such a low level where the Arduino IDE doesnt suit your needs anymore or a 328 would be the better choice.

eried, 2015-12-16:
Yeah, I am going to use a 328. Too much hustle. The tiny85 is just lovely, but not this time

matthijskooijman, 2016-07-03:
FWIW, I think the discussion above about supporting PCINT interrupts on all pins is orthogonal to the subject of this issue. The `digitalPinToInterrupt()` macro should apply for pins supported by the `attachInterrupt()` function, which I think the above proposals do. The PCINT interrupts are not supported by `attachInterrupt()`, but this applies equally to ATmega and ATtiny.