damellis/attiny #84: Invalid signature after trying to burn bootloader

LuisDiazUgena, 2015-12-02:
Hi!

I'm trying to burn the bootloader to make it run at 16Mhz with a external crystal and it always says the same error, although I had been able to upload the code before.

The verbose output of Arduino ide is [here](https://gist.github.com/LuisDiazUgena/0a87c583f9805e0f2214)

After trying to do burn the bootloader, I'm not able to upload the code, and [here is the output](https://gist.github.com/LuisDiazUgena/c44847a5bbdddf4982e9)

What can I do??

damellis, 2015-12-02:
Hmm.. in both cases the devices signature of the chip is reading as all zeros, which seems to suggest that the chip isn't responding properly. Is the external crystal hooked up? Are you sure it's hooked up correctly? (This seems like the kind of error you'd get if the chip was configured for an external clock but there wasn't a correct external clock attached.)

LuisDiazUgena, 2015-12-03:
That it was!!

Thank you so much!

closed 2015-12-03 by LuisDiazUgena