damellis/attiny #87: Software Serial requiring 8 MHz clock false or not?

deskwizard, 2016-01-11:
I would like some precision as per the validity of that statement for the following reasons:

1) Software Serial is working at 9600bps using 8MHz clock
2) 8 divided by 8 is 1.
3) 9600 divided by 8 is 1200.

As per those 3 statements, Software Serial works at 1200 bps using a 1 MHz clock.

And since this is all logic, here is number 4:

4) Tested working. Core v.1.0.1, IDE v1.6.7

I would just like to know if I am missing something obvious here as per the reason for that requirement/statement.

Thank you.
Charles Dorval (deskwizard)

deskwizard, 2016-01-11:
Also false: "And there’s no hardware serial port or I2C port" > it's called USI.

damellis, 2016-03-07:
SoftwareSerial used to not work at 1 MHz (because of a lack of a dedicated timing table at that speed). Now it does work at 1 MHz so I revised the tutorials. Thanks for pointing this out!

closed 2016-03-07 by damellis