damellis/attiny #89: Added 16MHz internal PLL clocks???

rtek1000, 2016-02-26:
Hello,

I would like to use Attiny85@16MHz with PLL,

https://recordnotfound.com/attiny-damellis-30720
show "Added 16MHz internal PLL clocks"

But IDE don't show this option
(Note: I'm using 1.6.5)

Thank you.

damellis, 2016-03-05:
We should support this, but I need to find a way to add it for just the ATtiny45/85 and not the ATtiny44/84 (which don't support it).

damellis, 2016-03-07:
This might require creating separate menu items in the top-level boards menu, one for the the ATtiny25/45/85 and a different one for the ATtiny24/44/84. I don't think there's a way for the selection in one sub-menu (i.e. Processor) to change the selections available in another sub-menu (i.e. Clock).

gioblu, 2016-03-10:
@damellis, ciao! You are right, there is no way to manage conditionally menu items.

So, a solution could be to have the menu structured this way:
attiny -> processor -> attiny85 -> 16 Mhz internal PPL clock
and remove the clock menu

This seems confortable for the user but adds overhead in the board file.
If you are ok with this solution I can send you the addon containing also the new menu structure.

gioblu, 2016-03-10:
Another user proposed a better solution here:
https://github.com/damellis/attiny/pull/86#issuecomment-194741256

Quote:
"Unless there's more logic control given by the menu declaration syntax,
I suspect the trick is to replace the menu hierarchy...

From:
|_ Processor
\_ Clock

To:
\_ Processor
    \_ Clock

So that the AtTiny85/45/25 get a different Clock menu naturally.

This might not bloat the boards.txt file too much...
It adds two clock menus, an nonPllClock and a pllClock."

NicoHood, 2016-06-21:
When I used The Attiny definitions here i found it very confusing that the attiny were not separated. So I suggest to add different categories for the x5 and x4 attinys anyways.

zaher, 2016-06-21:
For temporary, add it with label (attiny85 only)

closed 2016-07-02 by damellis

sonyhome, 2016-07-02:
Thanks for merging the change. We finally have 16MHz internal by default :P 

I saw you nuked all the json crap, which is good. Hopefully it still all works. However did you loose versioning? Does the new simpler packaging install and work fine?

damellis, 2016-07-02:
It worked for me. Let me know if you have any problems.

sonyhome, 2016-07-02:
I see...
* You did not declare the menu in same order as I did. I have a gripe that it always selects the slowest frequency by default, I find it kind of annoying to HAVE to fix both menus. Chances folks want 16MHz more than 1MHz most of the time.

* Did not use the same fuse values. I take it you want to keep other changes separate.

* The install guide still references the (now missing) json file. How does one install now?

gioblu, 2016-07-02:
Ciao @damellis thank you very much!!!! Finally!!!! :+1:

damellis, 2016-07-05:
@sonyhome: thanks for the comments!

I decided to leave the CPU entries in numeric order. This also has the advantage that the menu item defaults to the same value as the CPU factory settings.

I'm not necessarily opposed to other fuse changes but wanted to didn't want to hold up the PLL update until I had time to consider them. The best thing would be to submit separate pull requests for each change.

The JSON file is in another branch: https://github.com/damellis/attiny/tree/ide-1.6.x-boards-manager. If you see any references to other (incorrect) JSON files anywhere, please let me know so I can fix them.

Thanks again!

commit 6bba7d45 (2016-07-02): Add internal 16 MHz PLL clock support for ATtiny25/45/85.

Fixes #89