damellis/attiny #92: Does not work at all

Muessigb, 2016-03-07:
It shows attiny in the board menu and that's it.
No further selection.

damellis, 2016-03-07:
What version of Arduino are you using? How did you install the ATtiny support? Do you see Clock and Processor menus in Tools?

Muessigb, 2016-03-07:
Given that there is no readme, I have no idea what to do. I initially tried it with Arduino 1.0.5 and now updated to 1.6.7.

damellis, 2016-03-15:
There's a tutorial here: http://highlowtech.org/?p=1695. You may have better luck using the boards manager to install the ATtiny support. Otherwise, you probably need to move the ATtiny files into an avr sub-folder, so you have something like: ~/Documents/Arduino/hardware/attiny/avr/boards.txt, etc.

brunobasto, 2016-04-19:
+1 for having a README.MD. Could even be just a copy of what's in http://highlowtech.org/?p=1695

damellis, 2016-07-01:
Fixed by 78bf576cf5d4159f6957df24564476eb691f8734

closed 2016-07-01 by damellis