damellis/attiny #93: Compile error ATtiny85

dumindapium, 2016-03-15:
Hi,

I am getting this error for ATtiny85, and selected the processor, board from Tools menu.
installed this board from git zip to hardware folder.
Tried Arduino EDI versions 1.6.7 and 1.6.8(new).

Please advice.

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -hardware "C:\Users\Duminda\AppData\Local\Arduino15\packages" -hardware "D:\My-Projects\1Electronics\ARDUINO\Projects\hardware" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -tools "C:\Users\Duminda\AppData\Local\Arduino15\packages" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "D:\My-Projects\1Electronics\ARDUINO\Projects\libraries" -fqbn=attiny:attiny:attiny:cpu=attiny85,clock=internal1 -vid-pid=0X2341_0X0043 -ide-version=10608 -build-path "C:\Users\Duminda\AppData\Local\Temp\build3843495f2cbbf05c64008ec64242fd01.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "D:\My-Projects\1Electronics\ARDUINO\Projects\BlinkAtt85\BlinkAtt85.ino"
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -hardware "C:\Users\Duminda\AppData\Local\Arduino15\packages" -hardware "D:\My-Projects\1Electronics\ARDUINO\Projects\hardware" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -tools "C:\Users\Duminda\AppData\Local\Arduino15\packages" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "D:\My-Projects\1Electronics\ARDUINO\Projects\libraries" -fqbn=attiny:attiny:attiny:cpu=attiny85,clock=internal1 -vid-pid=0X2341_0X0043 -ide-version=10608 -build-path "C:\Users\Duminda\AppData\Local\Temp\build3843495f2cbbf05c64008ec64242fd01.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "D:\My-Projects\1Electronics\ARDUINO\Projects\BlinkAtt85\BlinkAtt85.ino"
""   -w -x c++ -E -CC -DF_CPU=1000000L -DARDUINO=10608 -DARDUINO_attiny -DARDUINO_ARCH_ATTINY         "-ID:\My-Projects\1Electronics\ARDUINO\Projects\hardware\attiny\attiny\cores\arduino" "-ID:\My-Projects\1Electronics\ARDUINO\Projects\hardware\attiny\attiny\variants\tiny8" "C:\Users\Duminda\AppData\Local\Temp\build3843495f2cbbf05c64008ec64242fd01.tmp\sketch\BlinkAtt85.ino.cpp" -o "nul"
""   -w -x c++ -E -CC -DF_CPU=1000000L -DARDUINO=10608 -DARDUINO_attiny -DARDUINO_ARCH_ATTINY         "-ID:\My-Projects\1Electronics\ARDUINO\Projects\hardware\attiny\attiny\cores\arduino" "-ID:\My-Projects\1Electronics\ARDUINO\Projects\hardware\attiny\attiny\variants\tiny8" "C:\Users\Duminda\AppData\Local\Temp\build3843495f2cbbf05c64008ec64242fd01.tmp\sketch\BlinkAtt85.ino.cpp" -o "nul"
""   -w -x c++ -E -CC -DF_CPU=1000000L -DARDUINO=10608 -DARDUINO_attiny -DARDUINO_ARCH_ATTINY         "-ID:\My-Projects\1Electronics\ARDUINO\Projects\hardware\attiny\attiny\cores\arduino" "-ID:\My-Projects\1Electronics\ARDUINO\Projects\hardware\attiny\attiny\variants\tiny8" "C:\Users\Duminda\AppData\Local\Temp\build3843495f2cbbf05c64008ec64242fd01.tmp\sketch\BlinkAtt85.ino.cpp" -o "C:\Users\Duminda\AppData\Local\Temp\build3843495f2cbbf05c64008ec64242fd01.tmp\preproc\ctags_target_for_gcc_minus_e.cpp"
exec: "-w": executable file not found in %PATH%
Error compiling for board ATtiny.

damellis, 2016-03-15:
I think the ATtiny files have to be in attiny/avr (not attiny/attiny) in the Arduino hardware directory. You might also try installing them with the boards manager, described here: http://highlowtech.org/?p=1695

dumindapium, 2016-03-16:
Thanks David,
I could install it using the boards manager as described http://highlowtech.org . This works fine.
Thanks for the board and the instructions.

closed 2016-03-16 by dumindapium