damellis/attiny #97: Warnings by overwritting of symbol "BIN" in "Print.h"

pediRAM, 2016-04-01:
I've reported the issue here: https://github.com/arduino/Arduino/issues/4784
Is it possible to rename the "BIN" symbol (in "iotnx4.h" etc.) to something else, so the usage of statements like "Serial.println(value, BIN)" or "lcd.print(value, BIN)" wouldn't output warnings or errors while compiling code for ATtiny X4?

NicoHood, 2016-04-01:
As I said its in the avr-libc. You cannot change this. You can close this issue. :)

pediRAM, 2016-04-01:
Oh! OK! Sorry!

matthijskooijman, 2016-07-03:
I guess the Arduino core could #undef BIN first to prevent the warning. Anyone care to submit this at https://github.com/arduino/Arduino/ ? It also occurs if you enable warnings in the IDE and select the Arduino Gemma board (which also uses an attiny).

NicoHood, 2016-07-03:
I've opened a PR some time ago already:
https://github.com/arduino/Arduino/pull/4791