damellis/attiny #98: attiny84 Support Arduino 1.6.5

Raynman77, 2016-04-01:
@damellis I'm having some issues with the attiny84. I'm using Arduino 1.6.5 and a Sparkfun AVR Tiny ISP programmer to flash the "rf24ping85.ino" to the attiny84 and changed the CE to #define CE_PIN 2 and CSN_PIN 3 in order to simply transmit from the attiny84 using a NRF24L01 radio to another NRF24L01 radio connected to an Arduino Pro Mini running the "Getting Started.ino". The attiny84 will not transmit.

I've run this same setup on an attiny85 and it works with no problem. The pins that I am using in my attiny84 sketch are correct but still no dice. I've also tried just a simple "Blink" sketch on the attiny84 to confirm that my pin definitions in my sketch are correct.

damellis, 2016-07-02:
Sorry, I'm not sure what to suggest here. Do you know what features of the pins are being used (e.g. PWM or interrupts or analog input)? That might explain differences between microcontrollers. 

In the meantime, I'm closing this, as it's not clear it's something we can fix in the ATtiny support itself.

closed 2016-07-02 by damellis

Raynman77, 2016-07-02:
Problem solved when I used a tiny core suggested by another user and also changed the pinout in the config file to meet my needs. It works now. Thanks.