damellis/attiny #102: recipe.preproc.macros pattern is missing

Rutherberg, 2016-06-01:
Hi,

When I try to compile for an attiny44 8Mhz internal, it returns 

"recipe.preproc.macros pattern is missing
Error compiling for board ATtiny."

Arduino IDE 1.6.9, Arduino Nano as ISP
Could you help me please ? I find nothing about this error related to the attiny with Arduino.

Thanks !

harishshankar, 2016-06-20:
even i had the same problem

damellis, 2016-07-01:
Weird, I don't see this... what version of the Arduino AVR boards do you have installed (in the Boards Manager)? What operating system?

AMKOSHESH, 2017-02-04:
even i had the same problem

Arduino: 1.8.1 (Windows 10), Board: "ATtiny13 (internal 9.6 MHz clock)"

The current selected board needs the core 'arduino:arduino' that is not installed.
Detecting libraries used...
recipe.preproc.macros pattern is missing

Error compiling for board ATtiny13 (internal 9.6 MHz clock).
The current selected board needs the core 'arduino:arduino' that is not installed.

Do you have a comment?

AMKOSHESH, 2017-02-05:
http://forum.arduino.cc/index.php?topic=453093.0

My problem was solved