damellis/attiny #105: Installing attiny breaks latest Arduino stable (1.6.10)

garrettr, 2016-07-27:
Arduino 1.6.10 was released [today](https://www.arduino.cc/en/Main/ReleaseNotes). After installing the Arudino 1.6.10 on Mac OS 10.10.5, I attempted to follow Arjun's AtTiny + Arduino Uno [tutorial](https://www.hackster.io/arjun/programming-attiny85-with-arduino-uno-afb829). The relevant steps of that tutorial are:

1. Add `https://raw.githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json` to `Additional Boards Manager URLs` in the Preferences.
2. Open Tools -> Board -> Board Manager. Search for `attiny` and **Install** the latest version (1.0.2).

Unfortunately, installing `attiny` breaks compiling (and not just for att!). To demonstrate this, open the blink example and attempt to compile it. It succeeds on Arduino 1.6.9, but fails on 1.6.10 with the error message `sorry - this program has been built without plugin support`.

<details>
<summary>Full error logs: Error compiling Blink after installing attiny</summary>
```
Arduino: 1.6.10 (Mac OS X), Board: "Arduino/Genuino Uno"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -hardware "/Users/garrett/Library/Arduino15/packages" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -tools "/Users/garrett/Library/Arduino15/packages" -built-in-libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/garrett/Documents/Arduino/libraries" -fqbn=arduino:avr:uno -ide-version=10610 -build-path "/var/folders/l8/stpbny9j3k1fxp9zjblsp5480000gn/T/buildd5ae82e677127ac286f51de66781d807.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/Applications/Arduino.app/Contents/Java/examples/01.Basics/Blink/Blink.ino"
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -hardware "/Users/garrett/Library/Arduino15/packages" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -tools "/Users/garrett/Library/Arduino15/packages" -built-in-libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/garrett/Documents/Arduino/libraries" -fqbn=arduino:avr:uno -ide-version=10610 -build-path "/var/folders/l8/stpbny9j3k1fxp9zjblsp5480000gn/T/buildd5ae82e677127ac286f51de66781d807.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/Applications/Arduino.app/Contents/Java/examples/01.Basics/Blink/Blink.ino"
Using board 'uno' from platform in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr
Using core 'arduino' from platform in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr
Build options changed, rebuilding all
Detecting libraries used...
"/Users/garrett/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/var/folders/l8/stpbny9j3k1fxp9zjblsp5480000gn/T/buildd5ae82e677127ac286f51de66781d807.tmp/sketch/Blink.ino.cpp" -o "/dev/null"
"/Users/garrett/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/var/folders/l8/stpbny9j3k1fxp9zjblsp5480000gn/T/buildd5ae82e677127ac286f51de66781d807.tmp/sketch/Blink.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/Users/garrett/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/var/folders/l8/stpbny9j3k1fxp9zjblsp5480000gn/T/buildd5ae82e677127ac286f51de66781d807.tmp/sketch/Blink.ino.cpp" -o "/var/folders/l8/stpbny9j3k1fxp9zjblsp5480000gn/T/buildd5ae82e677127ac286f51de66781d807.tmp/preproc/ctags_target_for_gcc_minus_e.cpp"
"/Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino10/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/var/folders/l8/stpbny9j3k1fxp9zjblsp5480000gn/T/buildd5ae82e677127ac286f51de66781d807.tmp/preproc/ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"/Users/garrett/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/var/folders/l8/stpbny9j3k1fxp9zjblsp5480000gn/T/buildd5ae82e677127ac286f51de66781d807.tmp/sketch/Blink.ino.cpp" -o "/var/folders/l8/stpbny9j3k1fxp9zjblsp5480000gn/T/buildd5ae82e677127ac286f51de66781d807.tmp/sketch/Blink.ino.cpp.o"
Compiling libraries...
Compiling core...
"/Users/garrett/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-gcc" -c -g -x assembler-with-cpp -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/wiring_pulse.S" -o "/var/folders/l8/stpbny9j3k1fxp9zjblsp5480000gn/T/buildd5ae82e677127ac286f51de66781d807.tmp/core/wiring_pulse.S.o"
"/Users/garrett/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/WInterrupts.c" -o "/var/folders/l8/stpbny9j3k1fxp9zjblsp5480000gn/T/buildd5ae82e677127ac286f51de66781d807.tmp/core/WInterrupts.c.o"
"/Users/garrett/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/hooks.c" -o "/var/folders/l8/stpbny9j3k1fxp9zjblsp5480000gn/T/buildd5ae82e677127ac286f51de66781d807.tmp/core/hooks.c.o"
"/Users/garrett/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/wiring.c" -o "/var/folders/l8/stpbny9j3k1fxp9zjblsp5480000gn/T/buildd5ae82e677127ac286f51de66781d807.tmp/core/wiring.c.o"
[... cut ...] 77127ac286f51de66781d807.tmp/core/new.cpp.o"
"/Users/garrett/Library/Arduino15/packages/arduino/tools/avr-gcc/4.8.1-arduino5/bin/avr-gcc-ar" rcs  "/var/folders/l8/stpbny9j3k1fxp9zjblsp5480000gn/T/buildd5ae82e677127ac286f51de66781d807.tmp/core/core.a" "/var/folders/l8/stpbny9j3k1fxp9zjblsp5480000gn/T/buildd5ae82e677127ac286f51de66781d807.tmp/core/wiring_pulse.S.o"
sorry - this program has been built without plugin support
exit status 1
Error compiling for board Arduino/Genuino Uno.
```
</details>

damellis, 2016-07-27:
Thanks for the report! The attiny install had dependencies on older versions of gcc and avrdude (relative to the ones included in 1.6.10). Installing them made all AVR compilations use them. This broke in the way you reported because that older gcc doesn't support plugins and/or link-time optimization (LTO), which 1.6.10 uses. 

For now, I removed the explicit tool dependencies from the attiny plugin. This should mean that it uses whatever version of avr-gcc and avrdude are built in to the Arduino IDE install.

You'll probably need to delete some / all of ~/Library/Arduino15 and then re-install the ATtiny plugin.

damellis, 2016-07-29:
I'm going to close this in the hopes that the change fixed the problem. If not, please let me know.

closed 2016-07-29 by damellis

zdanek, 2016-10-29:
I confirm that for 1.6.12 the problem no longer exists.

damellis, 2016-10-31:
Thanks @zdanek!