damellis/attiny #108: NeoPixelLibrary does not work on ATTiny45

SparkyCola, 2016-11-14:
I tried to run the simple example of [Adafruit's NeoPixelLabrary](https://github.com/adafruit/Adafruit_NeoPixel) on my ATTiny45 on every pin it doesn't work, no response at all (flashing via ArduinoISP worked well, blink example runs well and strip works  with UNO and Pro). I also tried to use the [light_ws2812,](https://github.com/cpldcpu/light_ws2812) and use `clock_prescale_set(clock_div_1);`  and capacitors/extra PSU and a pull-up-resitor for the data pin - still not working. Is there anything I can do about that?
I tried to run the simple example of [Adafruit's NeoPixelLabrary](https://github.com/adafruit/Adafruit_NeoPixel) on my ATTiny45 on every pin it doesn't work, no response at all (flashing via ArduinoISP worked well, blink example runs well and strip works  with UNO and Pro). I also tried to use the [light_ws2812,](https://github.com/cpldcpu/light_ws2812) and use `clock_prescale_set(clock_div_1);`  and capacitors/extra PSU and a pull-up-resitor for the data pin - still not working. Are there any suggestions about what could have gone wrong?

robhsaunders, 2016-11-23:
I'm having the same issue with the ATTiny84.  Just ran the Blink example and it runs fine.

robhsaunders, 2016-11-23:
I figured it out.  Using the StrandTest, once I uncommented the following lines it worked.  I did install a 16Mhz clock and burned boatloader with that setting.  Hope this helps.

   //This is for Trinket 5V 16MHz, you can remove these three lines if you are not using a Trinket
  #if defined (__AVR_ATtiny85__)
    if (F_CPU == 16000000) clock_prescale_set(clock_div_1);
  #endif
  // End of trinket special code