damellis/attiny #111: Clock Frequency does not get set

sonyhome, 2016-12-18:
With brand new installations of:
* Max OS X
* Arduino IDE 1.6.13
* AtTiny boards definition
* AtTiny85 chip

I noticed the chip's frequency does not actually get set.

To demonstrate I used an APA-102 LED strip with timed sequences and depending on the frequency they go faster or slower to the said ratio, instead of being the same speed. This explains why my bombproof demos were not working.

I'm going to investigate but if I recall this library is just a set of definitions, so the issue might be in the IDE.

sonyhome, 2016-12-18:
OK I'm an idiot. I remember now that the Arduino IDE does not actually reprogram the fuses of the chip, so whatever you choose on the IDE does not actually change the chip frequency. You just have to match the IDE choice to what the chip fuses are set to.

To set the fuses for AtTiny85:
https://dntruong.wordpress.com/2015/07/08/setting-and-reading-attiny85-fuses/

I'm gonna close this ticket if I can.

closed 2016-12-18 by sonyhome

damellis, 2016-12-18:
You can set the fuses using the "Burn Bootloader" command in the Arduino IDE. For this ATtiny core, there's no actual bootloader, but the command should still set the fuses.