damellis/attiny #116: Not working with Arduino IDE 1.8.x

erlanger, 2017-02-13:
The cores are not recognized with the Arduino IDE 1.8.x

Even though the board manager says they are installed. The error is the following:
```
Error compiling for board ATtiny25/45/85.
The current selected board needs the core 'arduino:arduino' that is not installed.

```

This happens AFTER adding `https://raw.githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json` to the preferences AND adding the Attiny board support in the board manager.

I presume 1.6.x works for IDE version 1.8.x  , let me know if that is not the case. Thanks for your work, I really appreciate it!

dunk8888, 2017-03-29:
There also a driver issue,may have to install the zedig driver libusbK i had to with 1.8.1 and 1.8.2

BrendanCode, 2017-05-30:
Maybe related to #117 but I noticed this problem after an update of Arduino IDE 1.8x

fabianofurtado, 2017-05-30:
I've got the same error here, using Arduino IDE 1.8.2 with Arch Linux 64 bits, avr-gcc 6.3.0 and arduino-avr-core 1.6.18.

yzubkov, 2017-07-17:
Same with Arduino IDE 1.8.3

Ted-CAcert, 2017-08-02:
Now I tried a new installation on another laptop, and there I did install the GitHub project in the sketchbook hardware folder, instead of using the board manager with https://raw.githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json

Now I get a similar error message:
`Die aktuell ausgewählte Board benötigt den Kern 'arduino:arduino', der nicht installiert ist.`

Maybe this helps?

isaacwellish, 2019-08-16:
Same here using Arduino 1.8.9. Found I got my sketch to upload to attiny 45 when going back to arduino 1.6.13