damellis/attiny #118: Wire.h fails

nedla2004, 2017-08-01:
I am using code similar to [this](https://www.arduino.cc/en/Tutorial/MasterReader) (minus Serial). All other tests have worked. The error message I get is quite long, but it is [here](https://pastebin.com/iB09pZq6).

Ted-CAcert, 2017-08-01:
Just ran into the same problem.

From what I found out, the ATtiny (at least the x5 variants, I don't know the x4 ones) supports TWI but implementation is quite different from the other AVR chips. While ATmega chips support a dedicated TWI module, ATtiny has a "Universal Serial interface" (USI), which can be used to implement TWI and SPI interfaces (and other things).

ATmega TWI and ATtiny USI use different registers, so the Wire library will fail to compile for ATtiny.

To me it looks like the low level part of the Wire library (twi.c and twi.h) would have to be re-implemented to support ATtiny.

Ted-CAcert, 2017-08-02:
After looking a bit more I found [http://playground.arduino.cc/Code/USIi2c](url)

There are several related repositories on GitHub, like [https://github.com/adafruit/TinyWireM](url) and [https://github.com/rambo/TinyWire](url) and several more. At this point I don't have an idea about which one is suited for which purpose.

Ted-CAcert, 2017-08-06:
FYI: The USIWire library worked well for my project, which has to fetch data from an Si7021 humidity sensor.

himijendrix24, 2019-10-14:
I just tried the adafruit TinyWireM library to use an attiny85 as a i2c-master and it did not work.
No compile-errors, but connection to slave never worked.

In the end i got it running by just changing this newer core:
https://github.com/SpenceKonde/ATTinyCore