damellis/attiny #127: Pin Defines for direct port. EG. PB1, PA3

jacky4566, 2019-01-25:
There seems to be a weird glitch with using pins by their port names. PB1, PA3.
When i run the following code. Only LED2 will blink.
I have verified the output works by specifying digital pin 3 but calling PB1 should also work no?
What is the exact function that handles this conversion?

```
const int LED1 = PB1;
const int LED2 = PA3;

void setup() {
pinMode(LED1, OUTPUT);
pinMode(LED2, OUTPUT);
}

void loop() {
digitalWrite(LED1, !digitalRead(LED1));
digitalWrite(LED2, !digitalRead(LED2));
delay(1000);
}
```

vgheo, 2019-01-25:
The physical pins on the chip can be addressed in (at least) two ways, corresponding to two abstraction levels :
1. Arduino level -   D0, D1, .... Dn
2. At ATtiny  register level - PA0...7, PB0..7, PC0..7

An arduino implementation defines a mapping from the Arduino pins to the chip registers.

For this  library ( damellis/attiny), the mapping is defined ( for the respective chips):
-  https://github.com/damellis/attiny/blob/master/variants/tiny8/pins_arduino.h
-  https://github.com/damellis/attiny/blob/master/variants/tiny14/pins_arduino.h

The pinout (including all possible functions of the pins) is detailed on [this diagram|https://cdn.instructables.com/FNU/75MM/IB8PT7B8/FNU75MMIB8PT7B8.LARGE.jpg].

Therefore - it is probable that the Arduino pin to register bit mapping that you assuming is not the one defined by the attiny library for the chip that you are using.

jacky4566, 2019-01-25:
Correct. 
Looking at the tiny14/pins_arduino.h file We can see both the subject pins are defined.
PB1 == D9
PA3 == D3

It doesnt seem weird that one works while the other doesnt?
```
digitalWrite(9, HIGH); //This works
digitalWrite(PB1 , HIGH); //This doesnt work

digitalWrite(3, HIGH); //This works
digitalWrite(PA3, HIGH); //This works
```

vgheo, 2019-01-26:
This instructable explains both alternatives 
https://www.instructables.com/id/ATTiny-Port-Manipulation/

I agree that [digitalWrite](https://github.com/arduino/ArduinoCore-avr/blob/5755ddea49fa69d6c505c772ebee5af5078e2ebf/cores/arduino/wiring_digital.c#L138) does incur performance and space penalties.

jacky4566, 2019-01-28:
Thank you vgheo for the clarification. The underlying API explanation is what i was looking for. 

I am aware of direct port manipulation. But i'm also a lazy programmer and dont want to write out port manipulation functions when they already exist like digitalWrite.  

I understand the concepts of Arduino and the ease of use they are trying to do with Digital and Analog pins but when getting more advanced or working with ATTINY its much easier to use the Pin name (PB1) directly instead of cross referencing back to the reference board. Just a personal preference. 

I see your reference to #8440 so we can consider this topic closed.

closed 2019-01-28 by jacky4566